sdr-enthusiasts/docker-adsb-ultrafeeder #18: looking for hints to improve my receiver setup

ebaschiera, 2023-05-31:
Hi,
I really enjoy playing with this bundle of feeders and the viewer. That's amazing.
I have a solid hardware setup, made of an AirNav usb stick and an antenna on the roof of my house. The software runs on a home lab pc, with enonugh cpu power and memory.
So I was wondering if someone can provide some hints and comments looking at my graphs... My biggest doubts are about gain. Unfortunately the autogain script isn't performing well in my setup, so I ended giving a try to several gain levels manually, and I chose the one that looked more promising. 
May I share here my graphs from graphs1090?
Thanks!

kx1t, 2023-06-18:
Sorry for the delay… yes you can share your graphs. 

Here’s some hints on how to manually optimize your gain from the graphs:

Here's how to hand-tune your ADSB SDR gain:
1. Look at the ADSB Signal Level graph in your Readsb graphs. It should look like the one attached:
- lowest signal level around -30 to -35 dB. If it's not, add or subtract from your gain so you get to this point.
- verify that the "green" area (2-4th quartile) is in the middle between the lowest signal and the max signal line
2. Then take a look at the text below the very first graph -- it contains the % messages > -3 dB. 
- In most cases, you want this to be between 2-5%. If it's 0, you can still add some gain without losing closeby planes due to oversteering the input. If it's larger than 5%, your gain is too high and you're basically losing closeby / strong signal gains

The images below are from readsb-protobuf and they may look slightly different with graphs1090 in ultrafeeder, but you get the idea:

[image]

[image]

ebaschiera, 2023-06-22:
Thanks for your help @kx1t !
With your explanation, I realized that my gain was very low. I had a bad experience with my previous USB dongle, which did not have a filter and the noise was too loud when the gain was high. With my current dongle (it has a filter) it looks like I can raise the gain level much more.
I will join your Discord server soon.
Thanks!

closed 2023-06-22 by ebaschiera