sdr-enthusiasts/docker-adsb-ultrafeeder #20: Migrating from readsb, remote SDR

FabioEight, 2023-06-14:
Hello,
My current setup (based on the old readsb gitbook guide), is made by two hosts, 1 PiZero 2W with RTL-SDR dongle doing only ADSB reception and 1 Debian retrieving all the feeder data and pushing it to providers.
Is this setup replicable using Ultrafeeder?
I don't see any parameters that I can use to disable processes not required on the PiZero SDR.
Also, it might be worth to add a section on GitBook, suggesting how to proceed with the upgrade from readsb to Ultrafeeder.
Thank you!

FabioEight, 2023-06-14:
So I tried to replicate my setup.
Here's my docker compose for my Pi Zero 2W (with RTL-SDR dongle):

```
  ultrafeeder:
    image: ghcr.io/sdr-enthusiasts/docker-adsb-ultrafeeder
    tty: true
    container_name: ultrafeeder
    hostname: ultrafeeder
    restart: always
    device_cgroup_rules:
      - 'c 189:* rwm'
    ports:
      - 8080:80
      - 30005:30005
    environment:
      # --------------------------------------------------
      # general parameters:
      - LOGLEVEL=error
      - TZ=${FEEDER_TZ}
      # --------------------------------------------------
      # SDR related parameters:
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_RTLSDR_DEVICE=${ADSB_SDR_SERIAL}
      - READSB_RTLSDR_PPM=${ADSB_SDR_PPM}
      #
      # --------------------------------------------------
      # readsb/decoder parameters:
      - READSB_LAT=${FEEDER_LAT}
      - READSB_LON=${FEEDER_LONG}
      - READSB_ALT=${FEEDER_ALT_M}m
      - READSB_GAIN=${ADSB_SDR_GAIN}
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      #
      # --------------------------------------------------
      # Sources and Aggregator connections:
      # Note - remove the ones you are not using / feeding
      #  Make sure that each line ends with a semicolon ";",  with the
      #  exception of the last line which shouldn't have a ";"
      - ULTRAFEEDER_CONFIG=
          adsb,feed.adsb.fi,30004,beast_reduce_plus_out;
          adsb,in.adsb.lol,30004,beast_reduce_plus_out;
          adsb,feed.adsb.one,64004,beast_reduce_plus_out;
          adsb,feed.planespotters.net,30004,beast_reduce_plus_out;
          adsb,feed.theairtraffic.com,30004,beast_reduce_plus_out;
          adsb,data.avdelphi.com,24999,beast_reduce_plus_out;
          mlat,feed.adsb.fi,31090,39000;
          mlat,in.adsb.lol,31090,39001;
          mlat,feed.adsb.one,64006,39002;
          mlat,mlat.planespotters.net,31090,39003;
          mlat,feed.theairtraffic.com,31090,39004
      # --------------------------------------------------
      - UUID=${ULTRAFEEDER_UUID}
      - READSB_FORWARD_MLAT_SBS=true
      #
      # --------------------------------------------------
      # TAR1090 (Map Web Page) parameters:
      - TAR1090_DISABLE=true
      #
      # --------------------------------------------------
      # GRAPHS1090 (Decoder and System Status Web Page) parameters:
      - GRAPHS1090_DISABLE=true
      #
      # --------------------------------------------------
    volumes:
      - /opt/adsb/ultrafeeder/globe_history:/var/globe_history
      - /opt/adsb/ultrafeeder/graphs1090:/var/lib/collectd
      - /proc/diskstats:/proc/diskstats:ro
[... 287 lines cut ...]
#    restart: always
#    devices:
#      - /dev/bus/usb:/dev/bus/usb
#    ports:
#      - 18082:8080
#      - 30005:30005
#    environment:
#      - TZ=${FEEDER_TZ}
#      - READSB_STATS_RANGE=true
#      - READSB_NET_ENABLE=true
#      - READSB_NET_CONNECTOR=192.168.1.9,30005,beast_in
#      - READSB_NET_ONLY=true
#    volumes:
#      - readsbpb_rrd:/run/collectd
#      - readsbpb_autogain:/run/autogain
#    tmpfs:
#      - /run/readsb
#      - /var/log
```

As you can see, it was a bit cleaner as there was the READSB_NET_ONLY parameter.
My current configuration is proper for my setup?
I'm currently unable to provide MLAT to the services.

Many thanks for your support.

kx1t, 2023-06-18:
Since you disabled graphs1090 and tar1090 on your pi zero receiver, ultrafeeder will run with minimum CPU footprint. You could probably make the startup time and memory footprint a bit better so by disabling things that really don’t matter on the Pi Zero, for example:

- no need to map port 8080:80 as there’s no web service
- Set `UPDATE _TAR1090=false`
- Set `TAR1090_ENABLE_AC_DB=false`
- No need to map any of these volumes as there’s no need to keep persistent data on the pi zero:
```yaml
      - /opt/adsb/ultrafeeder/globe_history:/var/globe_history
      - /opt/adsb/ultrafeeder/graphs1090:/var/lib/collectd
      - /proc/diskstats:/proc/diskstats:ro
      ``` 

for the rest, things look good at first sight

closed 2023-06-18 by kx1t

emes, 2023-11-15:
@FabioEight 

Other than wanting to use the same image, what was the motivation to change the Pi Zero?  Couldn't you continue to feed everything from the Debian server and just switch that to use Ultrafeeder?  Just curious...