sdr-enthusiasts/docker-adsb-ultrafeeder #21: ultrafeeder keeps shutting down

peteoheat, 2023-06-21:
I keep getting the below in the ultrafeeder logs and the container stays down.

`[2023-06-21 12:03:43.507][readsb] <3>High load: removeStale took 207/0 ms! stats: 0 (suppressing for 30 seconds)
[2023-06-21 12:05:30.566][readsb] [2023-06-21 12:05:30.475 BST] Caught SIGTERM, shutting down...
[2023-06-21 12:05:31.957][tar1090] tar1090.sh: exiting
sending SIGTERM to collectd
[2023-06-21 12:05:33.217][readsb] saving state .....
[2023-06-21 12:05:35.179][readsb]  .......... done, saved 602 aircraft in 1.962 seconds!
[2023-06-21 12:05:35.179][readsb] [2023-06-21 12:05:35.179 BST] Normal exit.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] 02-remove-prometheus-data: executing...
[cont-finish.d] 02-remove-prometheus-data: exited 0.
[cont-finish.d] 03-remove-mlathub-up: executing...
[cont-finish.d] 03-remove-mlathub-up: exited 0.
[cont-finish.d] 05-rtlsdr-biastee: executing...
[cont-finish.d] 05-rtlsdr-biastee: exited 0.
[cont-finish.d] 08-graphs1090: executing...
[cont-finish.d] 08-graphs1090: exited 0.
[cont-finish.d] done.
`

Any ideas?

kx1t, 2023-10-09:
Are you using HealthCheck/autoheal? It looks like something that would happen when your container goes unhealthy and autoheal is trying to restart it. 

You could check your container health with something like this:
```bash
docker inspect --format "{{json .State.Health }}" ultrafeeder | jq '.Lo
g[].Output' | awk '{gsub(/\\n/,"")}1' ```

kx1t, 2023-10-11:
Since this issue appears to be stale, I am going to close it. If you still have problems, please feel free to reopen it or to contact us on the Discord server (#adsb-containers channel) that is mentioned in the README of this repo for immediate help.

closed 2023-10-11 by kx1t