sdr-enthusiasts/docker-adsb-ultrafeeder #23: Recurring errors unable to connect to few services

hugalafutro, 2023-07-07:
Hi,
not sure if these services are down, or if my fw blocks the outgoing requests, but in ultrafeeder I always have this spammed every few seconds:
```
ultrafeeder  | [2023-07-07 09:40:48.609][mlat-client][mlat.planespotters.net] Connection to mlat.planespotters.net:39003 lost: [Errno 110] Connection timed out
ultrafeeder  | [2023-07-07 09:40:48.609][mlat-client][mlat.planespotters.net] Reconnecting in 0.5 seconds
ultrafeeder  | [2023-07-07 09:40:49.110][mlat-client][mlat.planespotters.net] Connection to mlat.planespotters.net:39003 failed: [Errno 99] EADDRNOTAVAIL
ultrafeeder  | [2023-07-07 09:40:49.111][mlat-client][mlat.planespotters.net] Reconnecting in 10.2 seconds
ultrafeeder  | [2023-07-07 09:40:53.057][mlat-client][feed.theairtraffic.com] Connection to feed.theairtraffic.com:31090 lost: [Errno 111] Connection refused
ultrafeeder  | [2023-07-07 09:40:53.057][mlat-client][feed.theairtraffic.com] Reconnecting in 11.1 seconds
ultrafeeder  | [2023-07-07 09:41:04.514][mlat-client][feed.theairtraffic.com] Connection to feed.theairtraffic.com:31090 lost: [Errno 111] Connection refused
ultrafeeder  | [2023-07-07 09:41:04.514][mlat-client][feed.theairtraffic.com] Reconnecting in 11.8 seconds
```

in the plane.watch feeder I recently noticed stopped working (but the webpage shows empty map too, so maybe a site issue?):
```
planewatch  | [pw-feeder] 2023-07-07T09:43:31+01:00 INF starting listener service dst=feed.push.plane.watch:12346 listen=127.0.0.1:12346 proto=MLAT
planewatch  | [pw-feeder] 2023-07-07T09:43:32+01:00 INF initiating tunnel connection to plane.watch dst=feed.push.plane.watch:12345 proto=BEAST src=ultrafeeder:30005
planewatch  | [pw-feeder] 2023-07-07T09:43:32+01:00 ERR ../../../../src/pw-feeder/pw-feeder/cmd/pw-feeder/stunnel.go:99 > could not connect error="dial tcp 110.173.227.69:12345: connect: connection refused" addr=feed.push.plane.watch:12345 name=BEAST
planewatch  | [pw-feeder] 2023-07-07T09:43:32+01:00 ERR tunnel terminated. could not connect to the plane.watch feed-in server, please check your internet connection error="dial tcp 110.173.227.69:12345: connect: connection refused" dst=feed.push.plane.watch:12345 proto=BEAST src=ultrafeeder:30005
planewatch  | [pw-feeder] 2023-07-07T09:44:00+01:00 INF connection established to local data source dst=feed.push.plane.watch:12346 listen=127.0.0.1:12346 proto=MLAT src=127.0.0.1:60438
planewatch  | [mlat-client] Fri Jul  7 09:44:00 2023 Connected to multilateration server at 127.0.0.1:12346, handshaking
planewatch  | [pw-feeder] 2023-07-07T09:44:01+01:00 ERR ../../../../src/pw-feeder/pw-feeder/cmd/pw-feeder/stunnel.go:99 > could not connect error="dial tcp 110.173.227.69:12346: connect: connection refused" addr=feed.push.plane.watch:12346 name=MLAT
planewatch  | [pw-feeder] 2023-07-07T09:44:01+01:00 ERR tunnel terminated. could not connect to the plane.watch feed-in server, please check your internet connection. error="dial tcp 110.173.227.69:12346: connect: connection refused" dst=feed.push.plane.watch:12346 listen=127.0.0.1:12346 proto=MLAT src=127.0.0.1:60438
planewatch  | [pw-feeder] 2023-07-07T09:44:01+01:00 INF starting listener service dst=feed.push.plane.watch:12346 listen=127.0.0.1:12346 proto=MLAT
planewatch  | [pw-feeder] 2023-07-07T09:44:01+01:00 ERR Could not bind to the requested tcp listening port, please check your configuration. error="listen tcp 127.0.0.1:12346: bind: address already in use" dst=feed.push.plane.watch:12346 listen=127.0.0.1:12346 proto=MLAT
planewatch  | [pw-feeder] 2023-07-07T09:44:02+01:00 INF starting listener service dst=feed.push.plane.watch:12346 listen=127.0.0.1:12346 proto=MLAT
planewatch  | [pw-feeder] 2023-07-07T09:44:02+01:00 ERR Could not bind to the requested tcp listening port, please check your configuration. error="listen tcp 127.0.0.1:12346: bind: address already in use" dst=feed.push.plane.watch:12346 listen=127.0.0.1:12346 proto=MLAT

```

I have followed the install instrustions to the t, and all the feeders were working ok (apart from the ultrafeeder spam with mlat.planespotters.net and feed.theairtraffic.com. I am currently feeding all the feeders form the guide book apart from RadarVirtuel which requires you to share personal data via email.

Should I ignore this?
Can I fix it on my side somehow?
Should I stop feeding plane.watch ?

Thanks for any info!

tvsjr, 2023-07-07:
I can't offer any help, but I experienced the same thing after rebuilding an older feeder with Ultrafeeder. My assumption is that the MLAT server for TAT is down - looking at their site, I can't even see feeder status, so perhaps they are having an outage.

I disabled plane.watch for the time being, assuming the same thing.

hugalafutro, 2023-07-08:
So the plane.watch site is back up and the container "autohealed", the main ultrafeeder container doesn't complain about `mlat.planespotters.net` so much anymore (few dropouts here and there), but it still complains about `feed.theairtraffic.com`
```
ultrafeeder  | [2023-07-08 08:25:12.748][mlat-client][feed.theairtraffic.com] Connection to feed.theairtraffic.com:31090 lost: [Errno 111] Connection refused
ultrafeeder  | [2023-07-08 08:25:12.749][mlat-client][feed.theairtraffic.com] Reconnecting in 10.9 seconds
ultrafeeder  | [2023-07-08 08:25:24.096][mlat-client][feed.theairtraffic.com] Connection to feed.theairtraffic.com:31090 lost: [Errno 111] Connection refused
ultrafeeder  | [2023-07-08 08:25:24.097][mlat-client][feed.theairtraffic.com] Reconnecting in 10.7 seconds
ultrafeeder  | [2023-07-08 08:25:35.050][mlat-client][feed.theairtraffic.com] Connection to feed.theairtraffic.com:31090 lost: [Errno 111] Connection refused
ultrafeeder  | [2023-07-08 08:25:35.050][mlat-client][feed.theairtraffic.com] Reconnecting in 11.3 seconds
```

hugalafutro, 2023-07-11:
Seems to have been fixed in the :latest image (from 20h ago). I just pulled it, rebuilt the stack and now it connects:
```
11/07/2023 11:13:05
[2023-07-11 11:13:05.530][mlat-client][feed.theairtraffic.com] Receiver: connected    85.0 msg/s received       26.1 msg/s processed (31%)
11/07/2023 11:13:05
[2023-07-11 11:13:05.530][mlat-client][feed.theairtraffic.com] Server:   ready         0.0 kB/s from server      0.3 kB/s to server
11/07/2023 11:13:05
[2023-07-11 11:13:05.530][mlat-client][feed.theairtraffic.com] Results:  20.9 positions/minute
```

closed 2023-07-11 by hugalafutro