sdr-enthusiasts/docker-adsb-ultrafeeder #38: [feature request] Allow selection of starting gain in autogain

andreagdipaolo, 2023-10-24:
I believe it could be useful to let the user select a starting gain in autogain.
In my current setup if I use autogain the system defaults at 49.6 and signals are so strong that they are detected as noise and I get something like 0,1% of what I would get with optimal gain.
I figured that the best gain for me is about 12-18.
I would still like to have autogain active in order to adjust for the environment etc etc, but that would be useful only if I could start at around 15 in order to get signals and make it work.

mclueppers, 2023-11-08:
This is already possible via [environment variables](https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder/blob/main/rootfs/etc/s6-overlay/scripts/readsb#L253).

Set in your `.env` file:
```
READSB_GAIN=autogain
READSB_AUTOGAIN_INITIAL_GAIN=15
```
and restart the container

andreagdipaolo, 2023-11-08:
Thanks, I had already solved that.
There is no reference about that string in the readme.
Anyway the script didn't work properly and @kx1t fixed it.
Thanks!

closed 2023-11-10 by kx1t