sdr-enthusiasts/docker-adsb-ultrafeeder #39: RTLs are "blanked" on container start

keatre, 2023-10-30:
I'm having trouble pinpointing what's going on here, but the basics are:

I have two RTL usb sticks for 1090 and 978 plugged into a Raspberry Pi 4 - when the pi boots, both RTL (which were serialized) display just fine. As soon as I start either ultrafeeder or dump978, the respective RTL then blanks out - which causes both docker images to say they cannot find an RTL that has the supplied serialnumber.

For instance, I can stop the docker service and see RTLs:

```
$ sudo rtl_sdr -d 0
Found 2 device(s):
  0:  AIRNAV, ADSB_1090, SN: 1090
  1:  Realtek, RTL2838UHIDIR, SN: 978

Using device 0: Generic RTL2832U OEM
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
```

However, as soon as I restart the docker service:

```
$ sudo service docker restart

$ sudo rtl_sdr -d 0
Found 2 device(s):
  0:  , , SN:
  1:  , , SN:

Using device 0: Generic RTL2832U OEM
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
```

I thought this might be a power thing, but I've tried both USB-C from an adequate power source, the POE+ hat, and even the RTLs on a powered USB hub that's connected to the pi - all have the same issue.

Any ideas as to what's going on?

fredclausen, 2023-10-30:
My best guess would be the kernel modules are either not unloaded or the UDEV rules aren't installed properly.

https://sdr-enthusiasts.gitbook.io/ads-b/setting-up-rtl-sdrs/blacklist-kernel-modules to blacklist the modules

and

```shell
sudo -E "$(which bash)" -c "curl -sL -o /etc/udev/rules.d/rtl-sdr.rules https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/osmocom-rtl-sdr.rules"
```

to install the UDEV rules. Restart the pi after those steps have been done.

keatre, 2023-10-30:
Unfortunately, I had run the standard install script that includes this - but I ran the steps again just to confirm.  It still seems to blank out each after docker starts.

Complete step:
```
[pi@piaware][~]
$ vcgencmd get_throttled
throttled=0x0

[pi@piaware][~]
$ adsb-usb
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 978
  1:  AIRNAV, ADSB_1090, SN: 1090

Using device 0: Generic RTL2832U OEM
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

[pi@piaware][~]
$sudo tee /etc/modprobe.d/blacklist-rtlsdr.conf <<TEXT1
# Blacklist host from loading modules for RTL-SDRs to ensure they
# are left available for the Docker guest.

blacklist dvb_core
blacklist dvb_usb_rtl2832u
blacklist dvb_usb_rtl28xxu
blacklist dvb_usb_v2
blacklist r820t
blacklist rtl2830
blacklist rtl2832
blacklist rtl2832_sdr
blacklist rtl2838

# This alone will not prevent a module being loaded if it is a
# required or an optional dependency of another module. Some kernel
# modules will attempt to load optional modules on demand, which we
# mitigate here by causing /bin/false to be run instead of the module.
#
# The next time the loading of the module is attempted, the /bin/false
# will be executed instead. This will prevent the module from being
# loaded on-demand. Source: https://access.redhat.com/solutions/41278

install dvb_core /bin/false
install dvb_usb_rtl2832u /bin/false
install dvb_usb_rtl28xxu /bin/false
install dvb_usb_v2 /bin/false
install r820t /bin/false
install rtl2830 /bin/false
install rtl2832 /bin/false
install rtl2832_sdr /bin/false
install rtl2838 /bin/false

TEXT1
# Blacklist host from loading modules for RTL-SDRs to ensure they
# are left available for the Docker guest.

blacklist dvb_core
blacklist dvb_usb_rtl2832u
blacklist dvb_usb_rtl28xxu
blacklist dvb_usb_v2
blacklist r820t
[... 105 lines cut ...]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)

[pi@piaware][~]
$ sudo service docker restart

[pi@piaware][~]
$ adsb-usb
Found 2 device(s):
  0:  , , SN:
  1:  , , SN:

Using device 0: Generic RTL2832U OEM
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        filename (a '-' dumps samples to stdout)
```

klara31, 2023-11-03:
do you have this in your docker-compose.yml, section for ultra feeder:
```
    devices:
      - /dev/bus/usb:/dev/bus/usb
```

keatre, 2023-11-03:
This has been solved

closed 2023-11-03 by keatre