sdr-enthusiasts/docker-adsb-ultrafeeder #41: where is mlathub for ultrafeeder?

tedder, 2023-11-10:
The examples here have entries for `mlathub`, but that container doesn't exist in the docker-compose. Should it? Or does ultrafeeder contain the mlathub? Here are the errors I'm seeing.

```
ultrafeeder    | [2023-11-10 18:36:23.729][mlathub] Beast TCP input: Name resolution for rbfeeder failed: Name or service not known
ultrafeeder    | [2023-11-10 18:36:23.729][mlathub] Beast TCP input: Name resolution for radarvirtuel failed: Name or service not known
ultrafeeder    | [2023-11-10 18:36:23.729][mlathub] Beast TCP input: Name resolution for planewatch failed: Name or service not known
ultrafeeder    | [2023-11-10 18:36:54.216][mlathub] Beast TCP input: Name resolution for piaware failed: Name or service not known
ultrafeeder    | [2023-11-10 18:36:54.216][mlathub] Beast TCP input: Name resolution for rbfeeder failed: Name or service not known
ultrafeeder    | [2023-11-10 18:36:54.216][mlathub] Beast TCP input: Name resolution for radarvirtuel failed: Name or service not known
ultrafeeder    | [2023-11-10 18:36:54.216][mlathub] Beast TCP input: Name resolution for planewatch failed: Name or service not known
```

dziban303, 2023-11-10:
Are you running rbfeeder, radarvirtuel, planewatch, and piaware containers in the same stack as ultrafeeder?

tedder, 2023-11-10:
Just running ultrafeeder like in [the example](https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder/blob/main/docker-compose.yml). Should I be running all the others?

Johnex, 2023-11-10:
Mlathub is built in, and if you don't use those containers above, you need to remove the lines for them :)

dziban303, 2023-11-10:
https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder#configuring-the-built-in-mlat-hub

If you aren't feeding piaware et al then you can remove those lines from your config.

tedder, 2023-11-10:
Aha, so the problem isn't that `mlathub` doesn't exist, it's that the extra containers for `piaware`, `rbfeeder`, etc don't exist.

closed 2023-11-10 by tedder