sdr-enthusiasts/docker-adsb-ultrafeeder #43: ultrafeeder service doesn't start after update

nyakove, 2023-11-16:
Hi! After update containers I have a cyclic error like 

`ultrafeeder  | [2023-11-16 14:56:00.049][readsb] WARNING -- READSB_FORWARD_MLAT_SBS has been set! Do not feed the SBS (BaseStation) output of this container to any aggregators!
ultrafeeder  | [2023-11-16 14:56:00.066][readsb] [s6wrap] !!! CAUTION !!! Wrapped program terminated by signal: 11 (Segmentation fault)
ultrafeeder  | [2023-11-16 14:56:00.066][readsb] [s6wrap] Command line for terminated program was: /usr/local/bin/readsb --net --quiet --lat 52.20896 --lon 20.92735 --uuid-file=/run/uuid --write-json=/run/readsb --write-state=/var/globe_history --json-trace-interval=15 --json-reliable=1 --net-ri-port=30001 --net-ro-port=30002 --net-sbs-port=30003 --net-bi-port=30004,30104 --net-bo-port=30005 --net-beast-reduce-out-port=30006 --net-json-port=30047 --net-api-port=30152 --net-sbs-in-port=32006 --heatmap-dir=/var/globe_history --heatmap=15 --net-sbs-reduce --forward-mlat-sbs --json-location-accuracy=2 --stats-range --net-api-port=30152 --net-beast-reduce-interval=1.0 --net-connector=dump978,30978,uat_in --net-connector=feed.adsb.fi,30004,beast_reduce_plus_out --net-connector=in.adsb.lol,30004,beast_reduce_plus_out --net-connector=feed.airplanes.live,30004,beast_reduce_plus_out --net-connector=feed.planespotters.net,30004,beast_reduce_plus_out --net-connector=feed.theairtraffic.com,30004,beast_reduce_plus_out --net-connector=data.avdelphi.com,24999,beast_reduce_plus_out --net-connector=skyfeed.hpradar.com,30004,beast_reduce_plus_out --net-connector=feed.radarplane.com,30001,beast_reduce_plus_out --net-connector=dati.flyitalyadsb.com,4905,beast_reduce_plus_out --net-connector=feed1.adsbexchange.com,30004,beast_reduce_plus_out --net-heartbeat=35 --db-file=/usr/local/share/tar1090/aircraft.csv.gz --device-type=rtlsdr --gain=49.6 --device=00000010 --ppm=30`

docker compose up / down doesn't help, restart device also. Can you please help, how I can fix it?

kx1t, 2024-01-02:
Is this still an issue? If yes, please provide the HW and OS environment in which you are deploying ultrafeeder. 
If it's no longer an issue, please feel free to close (or I will close the issue if I don't see any reaction)

Thanks!

nyakove, 2024-01-13:
@kx1t thanks, with the latest image update this issue is gone.

closed 2024-01-13 by nyakove