sdr-enthusiasts/docker-adsb-ultrafeeder #57: Network config

LLPP65, 2024-01-23:
First, thanks for this great project. When I was setting it up (ultrareader and several other feeders) I was getting random issues with problems connecting from readers to the ultrareader. What solved it was specifically adding the networking configuration to the docker compose file. Maybe this is something to add to the docs?

For each container I added:
    networks:
      - adsb_default

and at the end of the compose file I added:
networks:
  adsb_default:
    driver: bridge

Since then, all ok!

j-forristal, 2024-01-27:
Interesting feedback.  I am running a complex docker-compose for ultrafeeder (ultrafeeder + dump978 +airspy_adsb + planefence), and there is not a single explicit network config in there.  But it's all humming along just fine.  I'm on docker.io Ubuntu LTS package.

What OS flavor and container hosting package are you using?  Is this maybe a difference in containerd vs docker.io?

LLPP65, 2024-01-27:
After moving to Debian 12 (docker.io and docker compose plugin installed from docker.com package repository, v 25.0.1), it stopped working with the explicit network statements and works correctly without them. I was previously using Ubuntu 22.04, with docker.io and compose installed from the docker.com repository (not sure of that version).

kx1t, 2024-02-27:
Closing due to lack of activity - feel free to reopen if this is still an issue

closed 2024-02-27 by kx1t

wiedehopf, 2025-10-05:
Usually this goes via a docker default bridge network.
While they might also appear on the local DNS they typically work just fine.