sdr-enthusiasts/docker-adsb-ultrafeeder #59: Have SDR auto switch between 1090 and 978?

lg, 2024-01-26:
hey all, been running this docker container for about 2 weeks now, it's amazing. thank you for all the hard work put into such a complexity-necessitating system!

was wondering, is there any way to have the tuner switch between 1090 for ADS-B and 978 for UAT for lets say 10 seconds each? that way i wouldn't need to have two RTL-SDRs and two antennas, etc? i do understand the tradeoff of only getting "half" the data guaranteed, though i wonder if there are other tradeoffs. as a side note, i could also see functionality around tuners with wide band support catching both frequencies by centering between them.

sorry for being dense if this is obviously not possible. thx!

kx1t, 2024-01-26:
That's not something that we're planning to support. Extra SDRs are relatively cheap, and I'm sure that we'd spend 100s of hours trying to make this work - only to get to the conclusion that it doesn't work the way we'd want it.

Note that you can do 1090 MHz and 978 MHz reception on a single antenna. Yes, you'd lose a few dB in the splitter, but that shouldn't drastically affect your reception. For example - https://store.adsbexchange.com/products/5-5dbi-1090-978-antenna

Sorry - I'll close this issue as we unfortunately need to decline to support it.

closed 2024-01-26 by kx1t

dziban303, 2024-01-26:
I won't speak to the software feasibility side of this beyond saying it sounds like a lot of work for little benefit.

Unless one is right next to a busy general aviation airport, the amount of UAT traffic one is going to receive is like a single digit percent of the ADSB traffic, and probably a *lot* less for most locations. Also, swapping between frequencies would only work with a 'general purpose' SDR like the RTL-SDR Blog radios which lack filters or LNAs tuned for a given frequency. Most antennas, too, are (ideally) tuned to a specific frequency and aren't very efficient outside of that, though there are more broadband antenna designs, or at least, ones which claim they are but really just kind of suck for every band. Another issue is that shutting down the ADSB decoder and starting the UAT decoder and vice versa isn't an instant process, there will be several seconds of downtime; switching between bands every ten seconds would probably give you like six seconds of actual decoding.

There are no affordable SDRs that have a bandwidth of 112 MHz, which is what would be required to catch both UAT and ADSB messages. A typical RTL-based SDR has a bandwidth of about 2.5MHz. ~$100 SDRs can do, at most, 10MHz. A couple hundred bucks will get you like 40MHz. At that point, why not just buy another $20-$30 SDR so you can do both bands at the same time?

Another SDR is really the answer here. You could get away with using one antenna for both bands, using a $5 splitter going into two SDRs.

j-forristal, 2024-01-27:
I'm going to +1 what others said (sorry), but layer on more details not already mentioned and give you food for thought on what might turn out to be the better option.

First, if you make your SDR switch between 1090 and 978, you're taking MLAT (tracking military and stealthy planes that don't overtly broadcast their positions) off the table.  Since you are looking to UAT, I assume you are in the US.  And yes, US now requires most planes to have ADS-B or UAT, so MLAT relevance is minimizing -- but it's still relevant for the 'military cool stuff', which may or may not be what interests you. Hearing you want UAT, it seems to me you are looking to maximize your coverage -- and in doing so, this proposal would actually reduce your coverage.

Second, both the 1090 and 978 software do have some internal nuances that make some assumptions that they get to listen all the time.  The readsb ICAO internal cache "learns" valid aircraft as it goes, and later messages are accepted or rejected based on whether the app has confirmed/seen a different message from this aircraft previously.   Right now, there is no concept to these softwares to stop/shutdown but keep that learning across restarts... so even restarting the app makes it a blank slate and it can take a few seconds before it's back to fully learning the current aircraft population nearby.  Swapping every 10 seconds effectively gives it a lobotomy and it has to relearn everything flying around it, which can take a few seconds.

Which is a fancy way of saying: if you bounce back and forth between the two, you are going to have very degraded experiences with both.  Missing stuff galore, barely keeping up, and the worst of both worlds.  It's not the benefit you are looking for, it will not increase your coverage overall.

And that doesn't even touch on hardware.  To be able to bounce back and forth between 1090 and 978, you would need a generic/unfiltered SDR + (hopefully) an antenna that can do both frequencies.  Certainly possible, but in means you are probably still running a rig that is not operating to full potential, further putting you into a position to have degraded experiences for both.  Missing more planes, not getting more planes.

So, what's the best path forward?  I agree with the previous advisors: a US$8 bingfu (or similar) SMA coax-only splitter (to share your antenna) + US$30 second SDR is going to solve all the problems, give you the best of both worlds and not degrade either experience (minus some mild signal loss by using the splitter, but far less loss than the switcheroo setup you pitched).  If you already had a generic SDR, an extra US$5 will get you a filtered and amplified 1090 SDR, which will further enhance how much you pick up.  Move your existing generic SDR over to 978 use.

But, if you really really REALLY insist on this sharing concept, you can already do it with the docker containers and some cron magic: create crons that stop the 1090 container at certain times of the hour and immediately start a 978 container at the same time of the hour.  You'll have to read up on cron and how to script a basic docker-compose up/down every 10 minutes.  I guarantee you, after all that effort, the results are not going be satisfying.  And no, 10 seconds jumping back and forth makes no sense.  Give them 5-10 minutes before switching.

Overall, here's the jist: if your time is worth more than US$8/hour, the second SDR will probably be the cheaper route by far.  It's a bit more cost, but sooooo much more success.

Good luck on your tech journeys!

ps. depending upon your area, if you are near an airport with a primary radar, and the proximity of UAT aircraft to ADS-B aircraft, you might be getting TIS-B on 1090 that still shows you where they are.  I have a reciever that's 1090-only and it sees most the UAT aircraft--but I'm close to an international airport that runs a radar & rebroadcast to make that possible.  I assume that's probably not your case, if you are already on 1090 and asking for 978... but look at your map and see if you use TIS-B on your list.  You might already be covered.

lg, 2024-01-29:
thanks a lot y'all, really appreciate the insight. just trying to climb the charts of these radar websites while also keeping the complexity low. will get another SDR and look into either two antennas or a splitter. 👍