sdr-enthusiasts/docker-adsb-ultrafeeder #73: mlat-client errors in ultrafeeder log

baffles, 2024-03-29:
I see that I'm getting errors in my ultrafeeder log for every MLAT connection. Same "unrecognized arguments: ADS-B" error each time. This is with a mostly stock copy of docker-compose.yml from the gitbook.

```
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi] usage: mlat-client [-h]
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi]                    [--input-type {auto,dump1090,beast,radarcape_12mhz,radarcape_gps,radarcape,sbs,avrmlat}]
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi]                    --input-connect INPUT_CONNECT [--results RESULTS]
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi]                    [--no-anon-results] [--no-modeac-results] --lat LAT --lon
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi]                    LON --alt ALT [--privacy] --user USER [--server SERVER]
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi]                    [--no-udp] [--uuid-file UUID_PATH] [--uuid UUID]
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi]                    [--log-timestamps]
ultrafeeder    | [2024-03-29 13:49:23.845][mlat-client][feed.adsb.fi] mlat-client: error: unrecognized arguments: ADS-B
```

kx1t, 2024-03-29:
We had a bug in the parameter parsing code for a little bit, that should be fixed now. If it still doesn’t work, please post the exact value of the `ULTRAFEEDER_CONFIG` parameter here and I’ll take a look

baffles, 2024-03-30:
I did pull the latest earlier, and just pulled the latest again and still seeing this error (on `sha256:584049ba1ec55e32bd0379de0c76982aae9052fdc7a7a5fd888743046779ae24`).

Here's my ultrafeeder config:

```
               adsb,feed.adsb.fi,30004,beast_reduce_plus_out;
               adsb,in.adsb.lol,30004,beast_reduce_plus_out;
               adsb,feed.airplanes.live,30004,beast_reduce_plus_out;
               adsb,feed.planespotters.net,30004,beast_reduce_plus_out;
               adsb,feed.theairtraffic.com,30004,beast_reduce_plus_out;
               adsb,data.avdelphi.com,24999,beast_reduce_plus_out;
               adsb,skyfeed.hpradar.com,30004,beast_reduce_plus_out;
               adsb,feed.radarplane.com,30001,beast_reduce_plus_out;
               adsb,dati.flyitalyadsb.com,4905,beast_reduce_plus_out;
               adsb,feed1.adsbexchange.com,30004,beast_reduce_plus_out;
               mlat,feed.adsb.fi,31090,39000;
               mlat,in.adsb.lol,31090,39001;
               mlat,feed.airplanes.live,31090,39002;
               mlat,mlat.planespotters.net,31090,39003;
               mlat,feed.theairtraffic.com,31090,39004;
               mlat,skyfeed.hpradar.com,31090,39005;
               mlat,feed.radarplane.com,31090,39006;
               mlat,dati.flyitalyadsb.com,30100,39007;
               mlat,feed.adsbexchange.com,31090,39008;
               mlathub,piaware,30105,beast_in;
               mlathub,rbfeeder,30105,beast_in
```

baffles, 2024-03-30:
Actually, just noticed that `FEEDER_NAME` is used for the MLAT user value; which did have a space in it and end in "ADS-B" -- I changed my `MLAT_USER` value to something without a space and that seems to have fixed things. Wrapping it in quotes didn't help either. Not sure if spaces are to be avoided or if there's an issue handling that, but I'm and running now with a username without spaces.

closed 2024-04-30 by kx1t