sdr-enthusiasts/docker-adsb-ultrafeeder #75: remote SDR

Andrewhuk47, 2024-04-06:
Hi I have a SdrPlay RSP--1B that I have running on a VM. Is it possible to configure ultrafeeder to use that SDR with is accessed by the Vm's IP Address?

kx1t, 2024-04-30:
You would have to run [something like this](https://github.com/sdr-enthusiasts/docker-sdrplay-beast1090) on your VM. That would expose Beast data from your VM, which you can then access from any Ultrafeeder instance that can reach the VM.

Note that USB passthrough on VMs is often questionable - this may cause degradation, for example MLAT will often reject your data.

Generally it's better to not run your SDR receiver software components on a VM, but YMMV!

closed 2024-04-30 by kx1t

alkissack, 2025-02-03:
Andrewhuk47 - I got this working with:
      - ULTRAFEEDER_CONFIG=
        adsb,192.168.1.18,30005,beast_in

where 192.168.1.18 is a Pi running piaware with dongle attached.

Andrewhuk47, 2025-02-03:
Thanks for the comments. I have actually had it working for quite a while now. Just forgot about the post I made.