sdr-enthusiasts/docker-adsb-ultrafeeder #98: Lack of aircraft date/ownership

ofadam, 2024-06-10:
Despite having `TAR1090_ENABLE_AC_DB=true`, I'm only seeing the make and model of aircraft and no build year or owner registration. Any idea what I might need to do to gain that data?

dziban303, 2024-06-10:
those data are not present for most of the aircraft in the database, which is here: https://github.com/wiedehopf/tar1090-db/tree/csv

I believe it's only in the aircraft.json output anyway and doesn't appear in tar1090 web interface

ofadam, 2024-06-13:
The missing ownership data for aircraft does appear in /opt/adsb/globe_history/tar1090-update/aircraft.csv.gz – but even if setting `TAR1090_ENABLE_AC_DB=true` in .env, it doesn't seem to change anything (there's no difference when setting false vs true.)

wiedehopf, 2024-08-28:
```
readsb --help
[....]
  --db-file-lt                                                   aircraft.json: add long type as field desc, add field ownOp for the owner, add field year
```

kx1t, 2024-09-17:
Closing this issue -- if there's any new info, feel free to reopen

closed 2024-09-17 by kx1t

wiedehopf, 2024-09-17:
I'll elaborate, this parameter is needed on the readsb command line, it's not present in the container.

Adding those fields to the aircraft.json is disabled by default originally to keep aircraft.json a bit smaller, the whole DB stuff on readsb side was meant for really big instances.

Probably doesn't hurt to have --db-file-lt as a default param in the container ....