sdr-enthusiasts/docker-adsb-ultrafeeder #104: PROXMOX/DOCKER | FATAL: rtlsdr: error querying device #0: No such file or directory

einzelfabrik, 2024-06-19:
Hello everyone,
I have a special constellation - and am failing to pass the hardware through to the container.
About the setup: I have a **Proxmox Host VE 8.2.2** - the RTLSDR stick is connected to it. 
```
root@proxmox:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 004: ID 0bda:2832 Realtek Semiconductor Corp. RTL2832U DVB-T
```
I have an LXC container running with Docker + Portainer. I have adapted the config of the LXC container: 
```
lxc.cgroup2.devices.allow: c 189:* rwm
lxc.mount.entry: /dev/bus/usb/001/004 dev/bus/usb/001/004 none bind,optional,create=file
```
So the USB port(001-004) is available in the container - but how can I then pass it on to the ultrafeeder Docker container? 
All attempts ended in **`FATAL: rtlsdr: error querying device #0: No such file or directory` - can you give me tips?**

kx1t, 2024-06-29:
We generally don't provide support for ProxMox or container-in-container configurations (meaning: it may work but you're on your own); however:
- if you have made your USB port available to the ProxMox environment, then you can either use the `devices:` directive, or expose the `cgroup` directly to the Ultrafeeder container as shown [here](https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder/blob/main/docker-compose.yml) lines 11-12 and 109
- we have heard about issues with people having USB passthrough issues on ProxMox - best would be for you to ask on our [Discord server](https://discord.gg/km53D2ev) where there are other ProxMox users who may be able to assist

kx1t, 2024-09-17:
Closing this issue; if there is any additional info, feel free to reopen

closed 2024-09-17 by kx1t