sdr-enthusiasts/docker-adsb-ultrafeeder #140: mlat-client error in ultrafeeder log

hemna, 2024-09-11:
I am getting the following in my ultrafeeder container log.

```
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net] usage: mlat-client [-h]
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net]                    [--input-type {auto,dump1090,beast,radarcape_12mhz,radarcape_gps,radarcape,sbs,avrmlat}]
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net]                    --input-connect INPUT_CONNECT [--results RESULTS]
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net]                    [--no-anon-results] [--no-modeac-results] --lat LAT --lon
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net]                    LON --alt ALT [--privacy] --user USER [--server SERVER]
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net]                    [--no-udp] [--uuid-file UUID_PATH] [--uuid UUID]
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net]                    [--log-timestamps]
ultrafeeder  | [2024-09-10 21:41:55.738][mlat-client][mlat.planespotters.net] mlat-client: error: unrecognized arguments: ADSB
```

from my docker-compose.yml
```
services:
  ultrafeeder:
    image: ghcr.io/sdr-enthusiasts/docker-adsb-ultrafeeder
    # Note - if you want to enable telegraf for use with InfluxDB/Prometheus and Grafana,
    # use the following image instead:
    # image: ghcr.io/sdr-enthusiasts/docker-adsb-ultrafeeder:telegraf
    tty: true
    container_name: ultrafeeder
    hostname: ultrafeeder
    restart: unless-stopped
    device_cgroup_rules:
      - "c 189:* rwm"
    ports:
      - 8080:80 # to expose the web interface
      - 9273-9274:9273-9274 # to expose the statistics interface to Prometheus
    environment:
      # --------------------------------------------------
      # general parameters:
      - LOGLEVEL=error
      - TZ=${FEEDER_TZ}
      # --------------------------------------------------
      # SDR related parameters:
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_RTLSDR_DEVICE=${ADSB_SDR_SERIAL}
        #- READSB_RTLSDR_PPM=${ADSB_SDR_PPM}
      #
      # --------------------------------------------------
      # readsb/decoder parameters:
      - READSB_LAT=${FEEDER_LAT}
      - READSB_LON=${FEEDER_LONG}
      - READSB_ALT=${FEEDER_ALT_M}m
      - READSB_GAIN=${ADSB_SDR_GAIN}
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      #
      # --------------------------------------------------
      # Sources and Aggregator connections:
      # Notes - remove the ones you are not using / feeding
      #       - remove "adsb,dump978,30978,uat_in;" if you don't have dump978 and a UAT dongle connected to your station
      #       - !!! make sure that each line ends with a semicolon ";",  with the exception of the last line which shouldn't have a ";" !!!
      - ULTRAFEEDER_CONFIG=
        adsb,dump978,30978,uat_in;
        adsb,feed.adsb.fi,30004,beast_reduce_plus_out;
        adsb,in.adsb.lol,30004,beast_reduce_plus_out;
        adsb,feed.airplanes.live,30004,beast_reduce_plus_out;
        adsb,feed.planespotters.net,30004,beast_reduce_plus_out;
        adsb,feed.theairtraffic.com,30004,beast_reduce_plus_out;
        adsb,data.avdelphi.com,24999,beast_reduce_plus_out;
        adsb,skyfeed.hpradar.com,30004,beast_reduce_plus_out;
        adsb,feed.radarplane.com,30001,beast_reduce_plus_out;
        adsb,dati.flyitalyadsb.com,4905,beast_reduce_plus_out;
        mlat,feed.adsb.fi,31090,39000;
        mlat,in.adsb.lol,31090,39001;
        mlat,feed.airplanes.live,31090,39002;
        mlat,mlat.planespotters.net,31090,39003;
        mlat,feed.theairtraffic.com,31090,39004;
        mlat,skyfeed.hpradar.com,31090,39005;
        mlat,feed.radarplane.com,31090,39006;
        mlat,dati.flyitalyadsb.com,30100,39007;
        mlathub,piaware,30105,beast_in;
        mlathub,rbfeeder,30105,beast_in;
        mlathub,radarvirtuel,30105,beast_in;
        mlathub,planewatch,30105,beast_in
      # If you really want to feed ADSBExchange, you can do so by adding this above:
      #        adsb,feed1.adsbexchange.com,30004,beast_reduce_plus_out,uuid=${ADSBX_UUID};
      #        mlat,feed.adsbexchange.com,31090,39008,uuid=${ADSBX_UUID}
      #
      # --------------------------------------------------
      - UUID=${MULTIFEEDER_UUID}
      - MLAT_USER=${FEEDER_NAME}
           #
      # --------------------------------------------------
      # TAR1090 (Map Web Page) parameters:
      - UPDATE_TAR1090=true
      - TAR1090_DEFAULTCENTERLAT=${FEEDER_LAT}
      - TAR1090_DEFAULTCENTERLON=${FEEDER_LONG}
      - TAR1090_MESSAGERATEINTITLE=true
      - TAR1090_PAGETITLE=${FEEDER_NAME}
      - TAR1090_PLANECOUNTINTITLE=true
      - TAR1090_ENABLE_AC_DB=true
      - TAR1090_FLIGHTAWARELINKS=true
      #- HEYWHATSTHAT_PANORAMA_ID=${FEEDER_HEYWHATSTHAT_ID}
      #- HEYWHATSTHAT_ALTS=${FEEDER_HEYWHATSTHAT_ALTS}
      - TAR1090_SITESHOW=true
      - TAR1090_RANGE_OUTLINE_COLORED_BY_ALTITUDE=true
      - TAR1090_RANGE_OUTLINE_WIDTH=2.0
      - TAR1090_RANGERINGSDISTANCES=50,100,150,200
      - TAR1090_RANGERINGSCOLORS='#1A237E','#0D47A1','#42A5F5','#64B5F6'
      - TAR1090_USEROUTEAPI=true
      #
      # --------------------------------------------------
      # GRAPHS1090 (Decoder and System Status Web Page) parameters:
      # The two 978 related parameters should only be included if you are running dump978 for UAT reception (USA only)
      - GRAPHS1090_DARKMODE=true
      # - ENABLE_978=yes
      # - URL_978=http://dump978/skyaware978
      #
      # --------------------------------------------------
      # Prometheus and InfluxDB connection parameters:
      # (See above for the correct image tag you must use to enable this)
      #- INFLUXDBV2_URL=${INFLUX_URL}
      #- INFLUXDBV2_TOKEN=${INFLUX_TOKEN}
      #- INFLUXDBV2_BUCKET=${INFLUX_BUCKET}
      #- PROMETHEUS_ENABLE=true
    volumes:
      - /opt/adsb/ultrafeeder/globe_history:/var/globe_history
      - /opt/adsb/ultrafeeder/graphs1090:/var/lib/collectd
      - /proc/diskstats:/proc/diskstats:ro
      - /dev:/dev:ro
    tmpfs:
      - /run:exec,size=256M
      - /tmp:size=128M
      - /var/log:size=32M 
```      -

hemna, 2024-09-11:
Also seeing LOTS of these errors in the logs

```
[2024-09-10 21:53:08.514][nginx] 2024/09/10 21:53:08 [error] 551#551: *2 open() "/run/readsb/aircraft.pb" failed (2: No such file or directory), client: 192.168.11.1, server: _, request: "GET /data/aircraft.pb HTTP/1.1", host: "192.168.1.63:8080", referrer: "http://192.168.1.63:8080/script/readsb/backend/aircraftCollection.js"
```

kx1t, 2024-09-14:
Could you post the output of this:
``` 
docker logs ultrafeeder | grep mlat | grep planespotters | grep starting
``` 

feel free to obscure your lat/lon/uuid when you post here

it should look something like this, and there should be a strange “ADSB” piece in the startup string

```
[2024-09-13 12:45:04.041][mlat-client][mlat.planespotters.net] starting: /usr/bin/mlat-client --input-type auto --server mlat.planespotters.net:31090 --results beast,listen,39003 --results beast,connect,localhost:31004 --input-connect localhost:30005 --user xxxxxx --lat 42.xxxxxx --lon -71.xxxxx --alt 24m --uuid xxxxxxxx-xxxx-xxxx- xxx-xxxxxx
[2024-09-13 12:45:06.337][mlat-client][mlat.planespotters.net] mlat-client 0.4.2 starting up
```

wiedehopf, 2024-10-20:
`/run/readsb/aircraft.pb`
This is a file associated with readsb protobuf, not sure which software would be trying to get that from ultrafeeder.

Can you recheck with the current version if the errors are still present?

kx1t, 2024-11-14:
Closing due to inactivity -- if the issue still needs addressing, please feel free to reopen it. Thanks!

closed 2024-11-14 by kx1t