sdr-enthusiasts/docker-adsb-ultrafeeder #155: Docker container fails to start up - chmod /dev/console: read-only file system: unknown

acidicX, 2024-10-23:
I just upgraded my Pi to the latest arch linux ARM (aarch64) and I'm getting the following error messages  for ultrafeeder startup:

```
Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: chmod /dev/console: read-only file system: unknown
```

I'm using basically the docker compose setup from this repository.

If I comment out the line

#- /dev:/dev:ro

the container will start up, but won't find my rtlsdr device. I'm guessing something changed in the base Image you are using that now tries to chmod /dev/console which fails on RO filesystem, or a newer docker version is enforcing RO more strictly?

Is it really necessary to mount the whole `/dev` ?

Other containers start up fine on the host system. It's only ultrafeeder that fails currently.

Docker version:

> Docker version 27.3.1, build ce1223035a
> Docker Compose version 2.29.7

acidicX, 2024-10-23:
I've added a mount

```
- ./console-fake:/dev/console
```

to work around the issue for now

wiedehopf, 2024-10-25:
tty: true
if you remove that, does it work?
this was never needed.
/dev/bus/usb is sufficient to mount.

acidicX, 2024-10-29:
@wiedehopf without `tty: true` it works :)

I also changed the volume mount to

```
- /dev/bus/usb:/dev/bus/usb:ro
```

works as well!

Should I open a PR for the `docker-compose.yml` changes?

wiedehopf, 2024-10-29:
Not sure which ymls still have the tty.

```
- /dev/bus/usb:/dev/bus/usb
```
probably doesn't need ro for /dev/bus/usb

all the containers / yml which are doing rlt-sdr, /dev/bus/usb is the way to go
for serial stuff you need to mount /dev and need an extra line in the     `device_cgroup_rules:`

anyhow it's safe enough to change everything pretty much to 
```
    device_cgroup_rules:
      - 'c 189:* rwm'
...
    volumes:
      - /dev/bus/usb:/dev/bus/usb
 ```
 
 and removing tty: true is probably a good thing wherever you find it.
 
for non-rtlsdr stuff, probably fine to specify the necessary stuff in extra explanations.

acidicX, 2024-10-29:
https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder/pull/156

wiedehopf, 2024-10-29:
yeah the gitbook stuff also needs changes: <https://sdr-enthusiasts.gitbook.io/ads-b>

but that's a different story.

also this in the readme: <https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder/tree/main?tab=readme-ov-file#general-configuration>

anyhow you fixed the part that caused issues for you, ty :)

acidicX, 2024-10-29:
I added the README change to my PR!

acidicX, 2024-10-29:
and the gitbook PR https://github.com/sdr-enthusiasts/gitbook-adsb-guide/pull/160

closed 2024-11-04 by fredclausen

Johnex, 2024-12-17:
So this fix breaks the hot plug of usb. The only requirement to fix the issue is to remove the TTY: true lines from all the containers, no need to change the dev part. @fredclausen . I have tested this on my end, using the latest docker version and the same config since 1 year ago, only removing those lines.

@wiedehopf Also please for the future remember this when giving a suggestion, `/dev:/dev:ro`´ is required for hot plugging and fixing issues if a usb gets disconnected for any reason, removing that line is in most cases not a fix, and just a user that has not tested properly / understands what's going on.

PR #156 (2024-11-04): fix: remove tty and only mount usb subsystem
closes #155
commit be0ef15e (2024-11-04): fix: remove tty and only mount usb subsystem (#156)

closes #155
commit f3f00d40 (2024-11-04): fix: remove tty and only mount usb subsystem

closes #155