sdr-enthusiasts/docker-adsb-ultrafeeder #167: FR24 is not working properly and not feeding

amdjml, 2024-12-04:
I am using the ultrafeeder docker compose file. all other images working just fine except fr24. I see this in the logs for fr24

```
[2024-12-04 17:26:49.670][fr24feed] [time][e]Failed to synchronize time
[2024-12-04 17:26:49.670][fr24feed] [main][w]Failed to synchronize time, waiting 16 seconds
[2024-12-04 17:26:50.947][fr24feed] [reader][i]Connecting to unknown receiver via (tcp://ultrafeeder:30005)
[2024-12-04 17:26:50.960][fr24feed] [reader][e]Could not connect to tcp://ultrafeeder:30005
[2024-12-04 17:26:55.960][fr24feed] BeastBase::connectTcp(): Unable go connect, error: Network is unreachable[reader][i]Connecting to unknown receiver via (tcp://ultrafeeder:30005)
[2024-12-04 17:26:55.975][fr24feed] [reader][e]Could not connect to tcp://ultrafeeder:30005
[2024-12-04 17:27:00.975][fr24feed] BeastBase::connectTcp(): Unable go connect, error: Network is unreachable[reader][i]Connecting to unknown receiver via (tcp://ultrafeeder:30005)
[2024-12-04 17:27:00.988][fr24feed] [reader][e]Could not connect to tcp://ultrafeeder:30005
[2024-12-04 17:27:05.719][fr24feed] BeastBase::connectTcp(): Unable go connect, error: Network is unreachable[time][i]Synchronizing time via NTP
```

I have tried to do `docker compose pull` and it shows that I have the latest versions. I tried to recreate the docker containers and all other run just fine.

wiedehopf, 2024-12-04:
This is some bug in the fr24feed binary that probably has to do with DNS replies.

But i haven't been able to pin it down.
The binary is closed source after all.

If you want to provide some data, could you stop the container, run this:
```
sudo su -
apt install strace -y
while sleep 0.001; do strace -f -p $(pgrep fr24feed) > /run/fr24trace; done
```
Then start the container with it running.
After 30 seconds, interrupt it and upload /run/fr24trace using this for example:
```
curl -F'expires=24' -F"secret=" -F'file=@/run/fr24trace'  https://0x0.st
```

No clue if i can track this bug down further it's very weird.

amdjml, 2024-12-04:
when running the last command, I get this:

```
root@adsb:~# curl -F'expires=24' -F"secret=" -F'file=@/run/fr24trace'  https://0x0.st
451 Unavailable For Legal Reasonsroot@adsb:~
```

wiedehopf, 2024-12-04:
Heh, interesting, must be geoblocked (or tor).

```
cat /run/fr24trace | nc termbin.com 9999
```
This will work.
or you can probably upload it here if you can transfer it to your desktop first.

amdjml, 2024-12-04:
Here is the output:

[fr24.txt](https://github.com/user-attachments/files/18012648/fr24.txt)

wiedehopf, 2024-12-04:
Just for comparsion, which platform is this on?
amd64 or aarch64?

amdjml, 2024-12-04:
```
root@adsb:~# uname -a
Linux adsb 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

amdjml, 2024-12-04:
Thank you for looking into this. If I were to change the network to host and/or ultrafeeder, where do I make that change in the docker compose file and also changing the beast host to 127.0.0.1?

wiedehopf, 2024-12-04:
Yeah i'm not convinced this has anything to do with DNS / ipv6 anymore.
Just some super weird stuff of the binary not being able to connect.

Not sure how to debug this really without having access to the code.
Also motivation is limited.

Anyhow let me know if the workaround works for you.

wiedehopf, 2024-12-22:
This issue belongs with the fr24 container github, not this one.
Thus this issue can be closed.

kx1t, 2024-12-22:
Closing per @wiedehopf ’s advice

closed 2024-12-22 by kx1t