sdr-enthusiasts/docker-adsb-ultrafeeder #177: Using a locally modified tar1090 version

alkissack, 2025-02-03:
Trying to modify and/or use a local html source using "Using a locally modified tar1090 version".  but get errors in the logs or on the command line when running: docker exec -it ultrafeeder bash /tar1090-install.sh /run/readsb webroot /usr/local/share/tar1090 /var/tar1090_git_source

error:
[2025-02-03 11:29:18.952][tar1090-update] md5sum: images/sprites.png: No such file or directory
[2025-02-03 11:29:18.959][tar1090-update] mv: cannot stat 'images/sprites.png': No such file or directory
[2025-02-03 11:29:18.960][tar1090-update] [ERROR] Error in line 22 when executing: mv "$FILE" "$newname"
[2025-02-03 11:29:18.960][tar1090-update] [ERROR] Error in line 281 when executing: "$gpath/git/cachebust.sh" "$gpath/git/cachebust.list" "$TMP"

I see the images/sprite.png is in the orignal html/ but not in my repalcement, so the error is accurate, but I want ultrafeeder to just use my html/ folder on :8080

wiedehopf, 2025-02-03:
This is outside the scope of the feature, you're asking for a different feature.

closed 2025-02-03 by kx1t

wiedehopf, 2025-02-03:
See readme notes for CUSTOM_HTML please.

alkissack, 2025-02-03:
Thanks.  will have a read