sdr-enthusiasts/docker-adsb-ultrafeeder #188: What is the rationale behind this container?

sgofferj, 2025-03-23:
Not criticizing, but I'm a little bit confused. The primary philosophy behind containers is to have every service in separate, if possible ephemeral containers. What is the rationale behind having everything in one big container?

wiedehopf, 2025-03-23:
Ease of configuration.
It combines what most people are looking for.

The container still has the important stuff: packaging dependencies and the configuration in one place.

You could separate some stuff, but there isn't that much to gain by doing it.

sgofferj, 2025-03-24:
I see, thanks.

wiedehopf, 2025-03-25:
Which readsb container is probably the question.

readsb-protobuf is not the same readsb as used in this and the docker-tar1090 containers.

4G of memory is a ton.
ultrafeeder especially with no feeds and graphs disabled will use 20 MB i think.
I doubt it's an issue even with graphs, and you get nice system stats.

I prefer to run this on a pi:
GRAPHS1090_REDUCE_IO=true

Adds some more memory usage though, maybe 40 MB.
And if you disable graphs it's moot anyhow.

If you're memory constrained on a pi, i'd recommend this:
https://github.com/wiedehopf/adsb-wiki/wiki/zram-swap