sdr-enthusiasts/docker-adsb-ultrafeeder #195: latest-build-698: "libproc2.so.0: file too short" error

smkrv, 2025-04-18:
# Issue with ultrafeeder container on Orange Pi Zero 2W - "libproc2.so.0: file too short" error

## Environment
- Hardware: Orange Pi Zero 2W (aarch64)
- OS: DietPi v9.11.2
- Docker version: Docker Engine - Community, Version 28.1.1
- Ultrafeeder image: [latest-build-698](https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder/pkgs/container/docker-adsb-ultrafeeder/397893333?tag=latest-build-698)

## Problem Description
When running the ultrafeeder container, it starts normally but then repeatedly shows errors with the libproc2 library and s6-supervisor. The container appears to start its services but cannot function properly due to these errors.

## Error Logs
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service libseccomp2: starting
s6-rc: info: service 09-rtlsdr-biastee: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service libseccomp2 successfully started
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service startup: starting
s6-rc: info: service 09-rtlsdr-biastee successfully started
s6-rc: info: service startup successfully started
s6-rc: info: service timelapse1090: starting
s6-rc: info: service telegraf: starting
s6-rc: info: service tar1090-update: starting
s6-rc: info: service readsb: starting
s6-rc: info: service nginx: starting
s6-rc: info: service mlathub: starting
s6-rc: info: service mlat-client: starting
s6-rc: info: service graphs1090-writeback: starting
s6-rc: info: service collectd: starting
s6-rc: info: service cleanup_globe_history: starting
s6-rc: info: service aggregator-urls: starting
s6-rc: info: service adsbx-stats: starting
s6-rc: info: service timelapse1090 successfully started
s6-rc: info: service telegraf successfully started
s6-rc: info: service tar1090-update successfully started
s6-rc: info: service readsb successfully started
s6-rc: info: service tar1090: starting
s6-rc: info: service autogain: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service mlathub successfully started
s6-rc: info: service mlat-client successfully started
s6-rc: info: service graphs1090-writeback successfully started
s6-rc: info: service collectd successfully started
s6-rc: info: service graphs1090: starting
s6-rc: info: service cleanup_globe_history successfully started
s6-rc: info: service aggregator-urls successfully started
s6-rc: info: service adsbx-stats successfully started
s6-rc: info: service graphs1090 successfully started
s6-rc: info: service tar1090 successfully started
s6-rc: info: service autogain successfully started
s6-rc: info: service legacy-services: starting
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
s6-svc: fatal: unable to control /run/service/: supervisor not listening
s6-svc: fatal: unable to control /run/service/: supervisor not listening
s6-svc: fatal: unable to control /run/service/: supervisor not listening
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
145 (process ID) old priority 0, new priority 5
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
s6-svc: fatal: unable to control /run/service/: supervisor not listening
s6-svc: fatal: unable to control /run/service/: supervisor not listening
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
s6-svc: fatal: unable to control /run/service/: supervisor not listening
s6-rc: info: service legacy-services successfully started
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
s6-svc: fatal: unable to control /run/service/: supervisor not listening
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
s6-svc: fatal: unable to control /run/service/: supervisor not listening
s6-svc: fatal: unable to control /run/service/: supervisor not listening
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
ps: error while loading shared libraries: /lib/aarch64-linux-gnu/libproc2.so.0: file too short
s6-svc: fatal: unable to control /run/service/: supervisor not listening
/etc/s6-overlay/scripts/readsb: line 399: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/mlathub: line 109: /usr/local/bin/s6wrap: Success
/etc/s6-overlay/scripts/nginx: line 13: /usr/local/bin/s6wrap: Success
```

## Steps to Reproduce
1. Pull and run the ultrafeeder container
2. Check logs with `docker logs -f ultrafeeder`

## What I've Tried
- Tried multiple different builds of the container
- Tested several older versions of the image, systematically trying older builds
- After extensive testing of various versions, found that build-669 works correctly
- Cannot confirm if there are newer working builds between 669 and current, as I didn't test every single build
- Focused on testing versions with higher download counts
- Newer versions consistently show the "libproc2.so.0: file too short" error
- Attempted recreating the container multiple times with the latest image
- Cleared Docker cache and system resources before testing

## Additional Information
The issue seems related to a corrupted or incompatible library file for the aarch64 architecture. The container starts but cannot properly run the ps command or communicate with the s6 supervisor. 

Since build-669 works correctly while newer builds fail, this appears to be a regression in recent versions of the image for the aarch64 architecture, possibly related to updates in the procps package or its dependencies.

fredclausen, 2025-04-18:
Thanks for the report. We'll take a look at it. The only build change from the working builds to the current builds were an update in the debian base image we use for the containers, so likely something there is the culprit. I did some asking around in our discord server and no one else is seeing this (at least, as of now) which is curious.

dirkhh, 2025-04-18:
My guess is that something went wrong with the download of the baseimage layer that is shared between all of those recent builds - or, more likely, that when you wrote it to SDcard there was a bit flip somewhere.
I upgraded one of my test systems (an RPi4b - but runs the same exact binaries as a zero2) to ultrafeeder:latest-build-698 and cannot reproduce the issue. Specifically:
```
# docker exec ultrafeeder ls -l /lib/aarch64-linux-gnu/libproc2.so.0*
lrwxrwxrwx 1 root root     17 Dec 18  2022 /lib/aarch64-linux-gnu/libproc2.so.0 -> libproc2.so.0.0.1
-rw-r--r-- 1 root root 199112 Dec 18  2022 /lib/aarch64-linux-gnu/libproc2.so.0.0.1
```
Can you run the same command on your system (once you switched back to `latest-build-698`)

smkrv, 2025-04-19:
@fredclausen, @dirkhh, @wiedehopf, Thank you for the advice! 
The problem turned out to be more complex — it seems something was wrong with the file system in general. I completely reinstalled DietPi, rebuilt all Docker images, and everything worked properly. Unfortunately, pruning and complete cleaning of s2overlay docker didn't help, which is why I opened this issue. 

Everything is good now, my apologies for the false alarm and this issue, everything works! 

P.S.: Thank you for the awesome images and your work on them!

closed 2025-04-19 by smkrv